meshtastic/Meshtastic-Android #1726: [Feature Request]: Drop a pin

lewi3069, 2025-03-26:
### Contact Details

lewi3069@gmail.com

### Tell us your idea.

I'd like the app to be able to drop a pin on the map that is shared by DM or to the group. It would be awesome if you could set an age that way the pin could age off after say a week by default or even minutes or hours.

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

NomDeTom, 2025-03-26:
Have you tried waypoints?

Long press on the map, and fill in the details.

prokrypt, 2025-03-26:
> Have you tried waypoints?
> 
> Long press on the map, and fill in the details.

Would be nice to be able to select a target to send that waypoint to (everyone, specific channel, specific node, self)

todd2982, 2025-03-27:
> Would be nice to be able to select a target to send that waypoint to (everyone, specific channel, specific node, [self](https://github.com/meshtastic/Meshtastic-Android/issues/1640))

This follows a lot of the same methods as sending telemetry.  The waypoint is shared with your primary channel, and if it isn't sent as locked the people receiving the waypoint can adjust it.  You could set a private channel as your primary and move the default into another position.

The documentation on this feature is lacking at the moment.