meshtastic/Meshtastic-Android #1733: [Bug]: Seeing 1##/100 nodes when connecting to a device

AndrewReitz, 2025-03-30:
### Contact Details

aj.reitz@gmail.com

### What happened?

I'm assuming this is a bug, since numerator is larger than the denominator. If it's not feel free to close. I'm happy to take a look at the source and see if I can't figure out the issue or report more info here.

### App Version

2.5.21

### Phone

Pixel 8

### Device

heltec v3

### Firmware

2.6.4.b89355f Alpha 

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

jamesarich, 2025-03-30:
This might be an artifact of the fact that we're now retaining the app side nodedb across connections to the same device. 
@AndrewReitz is this being seen in the connection status text?

[image]

AndrewReitz, 2025-03-30:
@jamesarich yeah, I'm seeing something like 120/100 when connecting. Happy to take a screenshot or grab logs if those would help.

jamesarich, 2025-03-30:
actually, after checking the [source](https://github.com/meshtastic/Meshtastic-Android/blob/3bc4454c0e237988a3af5bb053c23fe50c8d7599/app/src/main/java/com/geeksville/mesh/service/MeshService.kt#L1577) - and some discussion with @b8b8 and @ianmcorvidae .. the `100`  that is being represented in here doesn't make a whole lot of sense, since a device's node db may actually be smaller or larger than 100 (as you have seen).

closed 2025-03-31 by jamesarich

AndrewReitz, 2025-03-31:
Makes sense! Thanks for the quick fix.

PR #1734 (2025-03-31): fix: Remove nodeDb size from status string
Previously the status string read `Nodes ( # / 100)` this updates the string to just show the raw node count `Nodes( # )` as a better representation, since a device's nodeDb may be smaller ( 80 in the case of nrf) or (200-250 for the cases of some esp32 devices with larger flash). 

resolves #1733