meshtastic/Meshtastic-Android #1742: [Bug]: Public Key Mismatch / Reset NodeDB doesn't clear list in app

jeremiah-k, 2025-03-31:
### Contact Details

Jack! on Discord

### What happened?

Everything was working well in 2.5.20, I'd been testing for a few days, but after upgrading to 2.5.21 I noticed Public Key Mismatches begin to occur.  At first it happened to one node and I was able to remove the node and the entry repopulated normally, as it should. 

Between switching nodes, I began to notice more of this occuring.  I tried resetting the nodedb but the nodes were still visible.  I checked with `meshtastic --nodes` and could see that the NodeDB was being reset properly.

I uninstalled and reinstalled the previous version (2.5.20). The issues disappeared.  

After reinstalling 2.5.21 the same issues reappeared, so there seems to be a regression in this latest alpha release.

### App Version

2.5.21 - Fdroid apk

### Phone

Pixel 6 Pro and Pixel 4a 5G both running GrapheneOS

### Device

Tbeam and T1000-E

### Firmware

2.6.4

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

garthvh, 2025-03-31:
The node db and the database in the app are working in conjunction to ensure that the PKC keys use for DM's are not compromised. A recent CVE has forced some changes to the android app to keep track of keys for nodes in the app that have potentially rolled out of the node db on device.  Because the node itself can only store a limited amount of nodes and keys the app can track additional key changes and alert the user to these changes.  

Connecting to different nodes, and clearing your node db is going to increase the amount of these key mismatch errors, the security vulnerability that is being addressed is that the android app and node were previously synced up, where now the app can store more nodes than the device.

Not clearing the nodedb all the time, and marking nodes you frequently communicate with as favorites will help make this a more seemless experience.

jamesarich, 2025-03-31:
@garthvh i do see something hinky going on - i don't think I caught all the cases where the app nodeDb is cleared. investigating.

jamesarich, 2025-03-31:
https://github.com/meshtastic/Meshtastic-Android/blob/d0816d34957ed04c6ce5881cd4295cba4d5947e3/app/src/main/java/com/geeksville/mesh/service/MeshService.kt#L429 i missed this the first time around, it looks like the db is still getting cleared in some cases - which could explain weird states

joneyes, 2025-04-04:
I'm seing the same thing android v2.5.21 and Portduino v2.5.18 (though just about to update this to >2.6)
Resetting the node DB from the android app or with --reset-nodedb will reset the database, but it looks like the node cache in the android app is not. It's repeatble.

garthvh, 2025-04-05:
This bug is in the beta track and is likely to remain for a few weeks.

brebtatv, 2025-04-06:
I confirm that version 2.5.20 is working and version 2.5.21 does not delete nodes and marks them in red as key mismatch.

joneyes, 2025-04-12:
Also now seeing public key mismatches against my own node now 😁

[image]

resunltd, 2025-04-18:
I am still having issues with nodes unable to send DM's to each other. It's been going on for at least 2 weeks. I can't seem to clear the issue and "start fresh."
All the nodes in question were working fine for months until I updated my Android App to 2.5.21
Reverted back to 2.5.20 and have been waiting a week for the errors to clear with no change. 
I get the Cloud with Slash immediately upon sending a DM
What can be done? Please advise. Thank you.

jamesarich, 2025-04-21:
this should be fixed by #1746 and in https://github.com/meshtastic/Meshtastic-Android/releases/tag/2.5.22

closed 2025-04-21 by jamesarich