meshtastic/Meshtastic-Android #1749: [Feature Request]: Conversation / chat muting or blocking

Astrodex-zz, 2025-04-03:
### Tell us your idea.

Currently in the conversation view I can see the default channel and my personal channel / chat. I don't want some devices to see any conversations from the default channel, but while I can mute it, the conversations still come through. It would be good to block / auto delete any messages coming in on any chat until re-enabled. I

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

b8b8, 2025-04-03:
Just remove the default channel and work with only your private channel. Keep a QR code handy for when you want re-add the the default channel.

madeofstown, 2025-04-08:
> Just remove the default channel and work with only your private channel. Keep a QR code handy for when you want re-add the default channel (that also includes your private channel).

If you do it this way, you'll probably want to make sure that your private channel uses the same channel slot as the default public one. Make a note of it before you delete the public channel, then set it back to that channel slot after and reboot the node.