meshtastic/Meshtastic-Android #1751: [Missing Translations]: Polish translation - "Favorites"

krzotr, 2025-04-05:
### What happened?

I ran the recent version of "Meshtastic Android 2.5.21 alpha". My language is set to **System default** (Polish). In #1520 favorites feature has been added. I see some missing translations.

[image]
[image]
[image]

### App Version

Meshtastic Android 2.5.21 alpha

### Phone

Google Pixel 8a android 15

### Device

LILYGO T-Beam V1.2

### Firmware

v2.6.4.b89355f

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

RCGV1, 2025-04-05:
You can translate those strings yourself, don’t need to know coding just make an account.
https://crowdin.meshtastic.org/

krzotr, 2025-04-05:
I added missing translations via crowdin.meshtastic.org

closed 2025-04-05 by krzotr