meshtastic/Meshtastic-Android #1753: [Feature Request]: Is there a reason why you still can't conect via domain on the mobile apps?

benjistokman, 2025-04-06:
### Tell us your idea.

Seems weird considering the web client is the exact opposite

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Firecul, 2025-04-07:
It would be a nice to have when you could set a local domain name `meshtastic.local` or `meshtastic.lan` but can't set a static ip (or can't remember it) for some reason.

madeofstown, 2025-04-08:
> It would be a nice to have when you could set a local domain name `meshtastic.local` or `meshtastic.lan` but can't set a static ip (or can't remember it) for some reason.

It would also be nice for using in a network environment that has a real domain.

slush0, 2025-05-15:
I second this. I have several Meshtastic nodes and remembering IP addresses for all of them is horrible UX.

OhaDerErste, 2025-05-19:
I would also like to see this implemented.

jamesarich, 2025-05-19:
Closing in favor of the more constructive duplicate #1834

closed 2025-05-19 by jamesarich