meshtastic/Meshtastic-Android #1760: [Bug]: pubkey mismatches on newly discovered nodes

shortwavesurfer2009, 2025-04-10:
### What happened?

I noticed nodes were giving me pub key mismatch errors and so I completely did a node DB reset and cleared all storage for the app and set it up completely fresh. Yet, despite doing that, newly discovered nodes are still coming up with pubkey mismatch errors.

[image]

### App Version

2.5.21

### Phone

InePlus Nord N200

### Device

SenseCAP t1000-e

### Firmware

2.6.1

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

thatSFguy, 2025-04-10:
NOTE:  I just found this bug as well.  If you disconnect and reconnect BLE, then the nodes reload and the red key (mismatch indicator) goes away.

shortwavesurfer2009, 2025-04-10:
Disconnecting from Bluetooth and reconnecting did not fix it for me. I even tried force stopping the app so that it was not running, then disconnected from Bluetooth, then reconnected to Bluetooth, and then restarted the app.

thatSFguy, 2025-04-10:
**Scenario:** Change in the app from node to "none", then back to the problem node
**Result:**  Key mismatch persists

**Scenario:** Change in the app to another node that is not turned, then back to the problem node
**Result:**  Key mismatch persists

**Scenario:** Change in the app to another node, let the app connect and nodes load, switch back to the problem node
**Result:**  Key mismatch disappears

garthvh, 2025-04-10:
Duplicate of https://github.com/meshtastic/Meshtastic-Android/issues/1742

closed 2025-04-10 by garthvh