meshtastic/Meshtastic-Android #1767: [Bug]: Text field almost hidden when keyboard enabled

SR7673, 2025-04-16:
### What happened?

Since i updated from 2.5.16 to 2.15.20 and higher (currently running 2.5.21, which is currently the most latest offered by F-Droid), the text field in which you type your message you want to send, is almost hidden as long as the keyboard is active on the phone.
So i can't see what i'm writing during typing the message. I need to disable the keyboard to see it before sending. And if there is a typo then it is some effort to correct it. I have to wite the message blindly!
The phone is an old Samsung Galaxy S5 but this worked until including 2.5.16...
Maybe it is just a matter of the display size and resolution?
F-Droid does not allow to downgrade to 2.5.16 :-(

### App Version

2.5.21

### Phone

Galaxy S5, Android 10

### Device

T-Beam 1.2

### Firmware

2.5.6

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

jsoberg, 2025-04-18:
Thanks @SR7673 - do you happen to have a screenshot of what you're seeing? Also wondering if you've modified your display size/font size in system settings at all

SR7673, 2025-04-19:
Thanks jsoberg. No, i never modified the display size. Screenshot attached...

[image]

jsoberg, 2025-04-19:
Thank you @SR7673! I believe this is related to https://github.com/meshtastic/Meshtastic-Android/issues/1603

PR #1771 (2025-04-21): fix: Remove imePadding modifier to internal scaffold box
fixes #1603
fixes #1767

`BaseScaffold` is applying `imePadding()` to an internal `Box` on all screen-level Composables. This is evidently unnecessary with our setup and applies double keyboard padding for older devices.

On my Samsung A10e, Android 9:
| Before | After |
|------|-----|
| <video src="https://github.com/user-attachments/assets/021e275c-0628-46b2-aa73-639ec5a4ed4c" width="300"></video> | <video src="https://github.com/user-attachments/assets/57737d25-4a3f-477c-be53-e1d7956e5a83" width="300"></video> |

Also tested on my Sam [... cut]
closed 2025-04-21 by jamesarich