meshtastic/Meshtastic-Android #1770: [Bug]: Changing LoRa defaults does not change frequency accordingly

b8b8, 2025-04-19:
### What happened?

I'm not sure if this is the android app or due to the firmware but when switching a node (over admin OR local bluetooth connection) from longfast to mediumfast the frequency does not update from 906.875 to 913.125.
The frequency remains at the previous 906.875 (this used to change). Use modem presets IS toggled on. 

### App Version

2.5.21

### Phone

Pixel 7a

### Device

RAK 4631

### Firmware

2.6.5

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

madeofstown, 2025-04-19:
I think if you have previously defined the channel slot number then it will not automatically change when you change the radio preset.

b8b8, 2025-04-19:
Setting it to zero and medium fast it reboots to 98...

[image]

b8b8, 2025-04-19:
Setting it to zero does not seem to change the to the correct default.

closed 2025-04-20 by b8b8