meshtastic/Meshtastic-Android #1773: [Bug]: Challen configuration menu deactivates "send" after first change

aghl100, 2025-04-20:
### Contact Details

andrew@aghl.org

### What happened?

When editing channels using the channel configuration menu on the Android app:
- after making the first set of changes (e.g. creating a new channel), the Save button lights up green, and can be clicked to send settings to the radio.
- after making further changes (e.g. adjusting the settings on that channel, deleting a channel) the Save button no longer lights up, and there is no way to send the changes to the radio.
- after coming out of the channel configuration menu and going back in, the Save button again works as normal.

Please could the Save button behaviour be modified so that it lights up green and is effective after every changes to settings rather than just the first?

### App Version

2.5.16

### Phone

OnePlus 8 pro, LineageOS 22.2

### Device

Heltec V3, also observed on a T3S3

### Firmware

2.6.4

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

madeofstown, 2025-04-21:
Typically settings changes cause the node to reboot, which causes it to disconnect from Bluetooth or WiFi. In order to make subsequent changes you need to reestablish the connection between the app and the node.

aghl100, 2025-04-21:
Many thanks for the response. 
I've just checked, and changing the channel settings isn't causing a reboot (I've checked the display doesn't go to "rebooting") and waiting doesn't make the "save" button re-enable after time for a reboot. I believe there's something else going on here.

aghl100, 2025-04-21:
(Apologies - marking as completed was an error!)

closed 2025-04-21 by jamesarich

aghl100, 2025-04-21:
Sorry, I believe that closing as "not planned" is in error. This is a bug report rather than a request for a new feature.,