meshtastic/Meshtastic-Android #1786: [Feature Request]: add key toggle to config import and export

b8b8, 2025-04-25:
### Tell us your idea.

Please add a toggle so keys can be separately imported or exported in the configs.

[image]

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

jamesarich, 2025-04-29:
@b8b8 it looks like this is only saving/restoring the `DeviceProfile` from https://github.com/meshtastic/protobufs/blob/master/meshtastic/clientonly.proto which currently does not contain the keys. We would need to add support for that - either add the keys to the `DeviceProfile` proto, or extend the client import/export config capability to include more than just `DeviceProfile`

jamesarich, 2025-06-18:
closed by #2158

closed 2025-06-18 by jamesarich