meshtastic/Meshtastic-Android #1787: [Bug]: Crash when clicking "More Details" on a node with a recently added HardwareModel

kabili207, 2025-04-26:
### Contact Details

Discord: kabi.chan

### What happened?

I have recently been working on a bot for my private MQTT broker that pretends to be an actual device on the network. While experimenting with this, I noticed that sending a NodeInfo packet with a HardwareModel that was added in version 2.6 of the protobufs will cause the Android app to crash when trying to view "More Details" on that node.

I originally tested this by broadcasting the April Fools `RESERVED_FRIED_CHICKEN` hardware model (because I was curious what would happen), but the crash also happens with hardware models added after `MESH_TAB`.

While it's perfectly understandable that it wouldn't know what to do with a bucket of fried chicken, the app really shouldn't crash if you happen to come across a recently released device in the wild and try to view more info on it.

### App Version

2.5.16

### Phone

Pixel 6, Android 15

### Device

RAK4631

### Firmware

2.5.20.4c97351

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

jamesarich, 2025-04-26:
Hi this should already be handled in the latest alpha builds. Feel free to give it a shot by joining the play store test group or selecting the latest on fdroid.

kabili207, 2025-04-26:
Tested with 2.5.21 from fdroid and it still crashes. Stacktrace from logcat is below

```
--------- beginning of crash
04-26 00:34:49.486 10703 10703 E AndroidRuntime: FATAL EXCEPTION: main
04-26 00:34:49.486 10703 10703 E AndroidRuntime: Process: com.geeksville.mesh, PID: 10703
04-26 00:34:49.486 10703 10703 E AndroidRuntime: java.lang.IllegalArgumentException: Can't get the number of an unknown enum value.
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at com.geeksville.mesh.MeshProtos$HardwareModel.getNumber(SourceFile:12)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at com.geeksville.mesh.model.MetricsViewModel.getDeviceHardwareFromHardwareModel(SourceFile:182)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at com.geeksville.mesh.model.MetricsViewModel.access$getDeviceHardwareFromHardwareModel(SourceFile:1)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at com.geeksville.mesh.model.MetricsViewModel$2.invokeSuspend(SourceFile:110)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at com.geeksville.mesh.model.MetricsViewModel$2.invoke(SourceFile:1)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at com.geeksville.mesh.model.MetricsViewModel$2.invoke(SourceFile:2)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at kotlinx.coroutines.flow.StartedLazily$command$1$1.emit(SourceFile:1)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at kotlinx.coroutines.flow.DistinctFlowImpl$collect$2.emit(SourceFile:579)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at kotlinx.coroutines.flow.FlowKt.emitAllImpl$FlowKt__ChannelsKt(SourceFile:130)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at kotlinx.coroutines.flow.FlowKt__ChannelsKt$emitAllImpl$1.invokeSuspend(SourceFile:12)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(SourceFile:9)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at kotlinx.coroutines.DispatchedTask.run(SourceFile:115)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at kotlinx.coroutines.EventLoopImplPlatform.processUnconfinedEvent(SourceFile:24)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at kotlinx.coroutines.internal.InlineList.resumeCancellableWith(SourceFile:142)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at kotlinx.coroutines.AbstractCoroutine.start(SourceFile:107)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at kotlinx.coroutines.JobKt.launch(SourceFile:24)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at kotlinx.coroutines.JobKt.launch$default(SourceFile:13)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at kotlinx.coroutines.flow.FlowKt.launchIn(SourceFile:8)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at com.geeksville.mesh.model.MetricsViewModel.<init>(SourceFile:123)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at com.geeksville.mesh.DaggerMeshUtilApplication_HiltComponents_SingletonC$ViewModelCImpl$SwitchingProvider.get(SourceFile:180)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at dagger.hilt.android.internal.lifecycle.HiltViewModelFactory$2.create(SourceFile:31)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at dagger.hilt.android.internal.lifecycle.HiltViewModelFactory.create(SourceFile:3)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at mil.nga.mgrs.utm.UTM$$ExternalSyntheticOutline0.$default$create(SourceFile:5)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at dagger.hilt.android.internal.lifecycle.HiltViewModelFactory.create(SourceFile:1)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at androidx.compose.ui.node.UiApplier.getViewModel$lifecycle_viewmodel_release(SourceFile:70)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at androidx.room.util.DBUtil.get(SourceFile:111)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at androidx.room.util.DBUtil.viewModel(SourceFile:3)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at com.geeksville.mesh.ui.NodeDetailKt.NodeDetailScreen(SourceFile:179)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at com.geeksville.mesh.navigation.NodeDetailNavigationKt$addNodDetailSection$1.invoke(SourceFile:9)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at com.geeksville.mesh.navigation.NodeDetailNavigationKt$addNodDetailSection$1.invoke(SourceFile:1)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(SourceFile:31)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(SourceFile:11)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at androidx.navigation.compose.NavHostKt$NavHost$32$1.invoke(SourceFile:1289)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(SourceFile:7)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at androidx.compose.runtime.AnchoredGroupPath.CompositionLocalProvider(SourceFile:77)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at androidx.compose.runtime.saveable.SaveableStateHolderImpl.SaveableStateProvider(SourceFile:151)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at androidx.tracing.Trace.access$SaveableStateProvider(SourceFile:116)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at androidx.navigation.compose.NavHostKt$NavHost$32$1.invoke(SourceFile:42)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(SourceFile:7)
[... cut ...]
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:591)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:911)
04-26 00:34:49.486 10703 10703 E AndroidRuntime: 	Suppressed: kotlinx.coroutines.internal.DiagnosticCoroutineContextException: [StandaloneCoroutine{Cancelling}@2b77c32, Dispatchers.Main.immediate]
04-26 00:34:49.489  1668  2263 W ActivityTaskManager:   Force finishing activity com.geeksville.mesh/.MainActivity
```

b8b8, 2025-04-26:
https://github.com/meshtastic/Meshtastic-Android/releases/tag/2.5.22

kabili207, 2025-04-26:
Same error with 2.5.22

jamesarich, 2025-04-26:
Thanks for checking.

closed 2025-04-27 by jamesarich

kabili207, 2025-04-28:
Confirmed that it no longer crashes with 2.5.23. Thank you very much!

PR #1789 (2025-04-27): fix(#1787): handle unknown hardware details crash
Handle potential exceptions when decoding `device_hardware.json` or the proto.

Refactors how we store/load the svg images.
Updates svg images and device_hardware.json to latest versions available from web flasher.

fixes #1787
fixes #1762