meshtastic/Meshtastic-Android #1790: [Bug]: "More options" menu not accessible

true911, 2025-04-26:
### Contact Details

true911m@gmail.com

### What happened?

I'm trying to delete a node from my local list.  I long-press the menu button and I see "more options". I press on that and it disappears, without presenting any options at all.

### App Version

2.5.16

### Phone

Motorola Edge (2022), Android 14

### Device

T1000-E, multiple RAK 4631 devices

### Firmware

2.6.4.b89355f beta on 4631

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

true911, 2025-04-26:
I also tested this on a Pixel 6 running Android 13.  Identical behavior.

b8b8, 2025-04-26:
Upgrade the app to beta or alpha and retry. Latest version is 2.5.22

true911, 2025-04-27:
Both alphas, "Something went wrong.  App not installed."

If the bug has been addressed, just let me know and I'll wait until it makes it to the Play Store.

true911, 2025-05-01:
> Nodes can be removed by tapping on their shortname in their list item and
> tapping "Remove" - are you saying that the general "more options" menu item
> on the toolbar doesn't open a menu for you?
>
> Attached video to describe the removal process, and tapping/long-pressing
> on the menu:
>
> https://github.com/user-attachments/assets/903ae165-41b3-4f04-9943-b9996f0ccbe0

That is what I am saying, yes.

I am not trying to remove a node from the "nodes" page, i.e. nodes that
have been seen on the radio.  I am trying to remove one of my personal
configured nodes from the configuration page (the far-right gear icon),
because it will not update its name on that screen after I renamed it.  I
thought I could remove/forget it from that list and add it again from
scratch with the proper name displayed.

Someone on youtube or elsewhere said to use that long-press menu to remove
it, but it doesn't show me any menu options.
__________________________________________________

*Mark ***@***.***

b8b8, 2025-05-01:
remove it by forgetting the bluetooth connection

true911, 2025-05-01:
That's what I want, do I do that from the bluetooth menu?

b8b8, 2025-05-01:
from your phone  bluetooth pairing list, remove the offending node, not from within the meshtastic app. you are wanting to delete a bluetooth pair.

true911, 2025-05-01:
The original post, however was about the long-press menu not appearing.  If I long-press the 3 dots, I get the "More Options" button, but when I touch that it just disappears.  No menu afterwards.

closed 2025-05-01 by jamesarich