meshtastic/Meshtastic-Android #1808: [Feature Request]: Expose more text elements to the node list filter

b8b8, 2025-05-01:
### Tell us your idea.

I think it would beneficial to expose the following elements to the already in place text filter:
1. Hardware (heltec-v3, station-g2 etc.)
2. Role (client, router_late etc.)
 

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

shalberd, 2025-05-04:
is_licensed would be cool to have, too, in GUI in general and in filter, see https://github.com/meshtastic/Meshtastic-Android/issues/1808

b8b8, 2025-05-04:
Standard meshtastic traffic is encrypted. Ham traffic is illegal to be encrypted (in most places). The projects stance is that these two networks shall not mix. You should not have any licensed  nodes on meshtastic network.

shalberd, 2025-05-05:
> Ham traffic is illegal to be encrypted (in most places). The project's stance is that these two networks shall not mix. You should not have any licensed nodes on the standard meshtastic network.

@b8b8 Well, in that case, I do not understand why, in EU band EU868, you even offer, under Radio Configuration - User,
the option "Licensed amateur radio".
is_licensed / **HAM mode, should never ever be allowed when a band EU868 is selected under Lora Config**. Since EU868 is not an official HAM radio / amateur radio band, as opposed to EU433.
Currently, this has led to rogue nodes, having enabled is_licensed, on our LongFast network here. With no way to quickly filter for the nodes that have that mode enabled.
So, any way you cut it, and I know there have been discussions of this before, this current state of affairs is extemely bad.
Honestly, I don't even understand why is_licensed / HAM mode is offered at all. This is not the spirit of Meshtastic as a project.

b8b8, 2025-05-05:
The ham licensee took the test, has their address registered, knows the rules, and is transmitting their call-sign. 
If they are operating illegally, that's on them and they could still pick any other band close by and override the frequency to EU868.

b8b8, 2025-05-07:
@shalberd if the standard meshtastic network and a ham meshtastic network don't mix what would be the point of the text filter?

shalberd, 2025-05-08:
@b8b8 I am running firmware 2.6.4 and the latest Android App version. I am on a default, LongFast Channel 0 network, and I see nodes with is_licensed: true in the app debug log, I see those nodes in the node list / node DB as well. 
So to me, this means that the changes you mention above / link are a good step, but something else is missing, as clearly there are nodes with is_licensed: true status on the EU868 LoRa network. There is a mix still. Best would be if there would at least be some warning in the Web config application and the apps in radio config - Lora is someone set both is_licended / licensed amateur radio user as well as EU868, as that channel clearly is not an amateur radio channel in Europe. See also the discussion at https://github.com/meshtastic/firmware/discussions/6378
As it is now, there is no way for me to filter for those nodes in the app's node list.

closed 2025-06-15 by github-actions[bot]