meshtastic/Meshtastic-Android #1810: [Bug]: Android BLE confuses name and short-name

roberthadow, 2025-05-01:
### Contact Details

Discord Jumper

### What happened?

When one seeks to pair a Heltec V3 with an Android phone, the advertised device name is changed from the normal concatenation of "Meshtastic_" and last four characters of the MAC address to a concatenation of the owner-short, then underscore, then last four characters of the MAC address. 

Selections on my phone:

*  None (disable)
*  9BF0_9bf0 [see note 1 below]
*  BB38_bb38
*  Meshtastic_b4f8  [see note 2 below]
*  Meshtastic_ce44
*  192.168.1.201
*  192.168.1.202
*  192.168.1.200
*  192.168.1.211
*  192.168.1.203
* */dev/bus/usb/001/013

Note 1:  There is no "9BF0_9bf0".  The registered owner is "Dover Relay 9BF0"

Note 2: There is no "Meshtastic_b4f8". The registered owner is "Denville Relay B4F8"

### App Version

2.6.4

### Phone

Motorola Moto G Play 2023; Android 13

### Device

Heltec V3

### Firmware

2.6.4

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

garthvh, 2025-05-01:
Once you change the short name the Bluetooth name is the short name and last 4 characters of the MAC address, this is created by the firmware

closed 2025-05-01 by garthvh

jamesarich, 2025-05-01:
Also @roberthadow - if you update the short name, forgetting the bt device (via android system bt settings) and repairing (via the app) will show the new short name.

roberthadow, 2025-05-01:
Having to (1) update a owner-short, (2) forget a BLE device, then (3) a repair, is a workaround but not a solution.

I umambiguously define attributes in these two ways: 

`meshtastic --host 192.168.1.211   --set-owner-short 'CE44' --set-owner 'Meshtastic ce44'

meshtastic --host 192.168.1.201 \
    --set-owner-short "B4F8" \
    --set-owner "Denville Relay B4F8" `

I believe I should be able to rely on the configuration settings.  In situtations that I cannot yet identify, the owner-short reverts to lower case without a user command.

In any case, I appreciate your quick response.

Robert

jamesarich, 2025-05-01:
> Having to (1) update a owner-short, (2) forget a BLE device, then (3) a repair, is a workaround but not a solution.

Unfortunately this information is stored in the android system bluetooth device profile, which we have little/no control over so a workaround is needed.

garthvh, 2025-05-01:
> Having to (1) update a owner-short, (2) forget a BLE device, then (3) a repair, is a workaround but not a solution.
>
> I umambiguously define attributes in these two ways:
>
> `meshtastic --host [...]

These are security features on both mobile OS's to prevent BLE spoofing so the functionality that lets BLE name change would have to be removed to prevent the need to forget devices.  I do think that android should stop automatically generating a short name when long name changes.