meshtastic/Meshtastic-Android #1819: [Feature Request]: show info about node HAM mode is_licensed true false in Node Details

shalberd, 2025-05-04:
### Contact Details

sven.thoms@gmail.com

### Tell us your idea.

I'd like the app to show the node info field is_licensed when clicking, in the node list, on a node's short name - More Details

in addition to role, hardware, last heard, the GUI should also show, under Device

Licensed amateur radio: true / false

the information on that lies in the nodeInfo is_licensed field.

What do you think?

We have some users here in Europe EU868 that have is_licensed enabled even though it is not even an official amateur radio band ... so at least seeing that in the GUI would be great. Filtering for that in node list would be awesome, too.

Some time in future, it would be nice if licensed amateur radio mode enablement in the GUI were only possible in EU433, not EU868 ...

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Miaoucat0, 2025-05-27:
personally i think the whole Ham Radio setting is confusing. if you dont want to use encryption for whatever reason then stay on the default channel and thats it. **its the responsability of the user to ensure he is using material (radios) and frequencies that are legal in his country in the first place.** i dont know about the ISM band/Ham radio band overlap in other countries*, but until the program can be set to tailor its frequency, power limit and even datarate/bandwith to the specific legal limits in various countries then it is not very useful. (the Ham setting)

*i just checked and the 433-915 bands are basically solely for europe and the americas respectively. 

[image]

i am actually a licensed Amateur Radio Operator in Canada and member of my country Civil Defence organization. maybe i am missing the point of the Ham setting? from my understanding it is just removing encryption and calling out your callsign at X interval. it should be alright just to use the default LongFast channel? since everybody has that key i woudnt think it be considered as trying to hide your transmissions.

shalberd, 2025-05-27:
@Miaoucat0 I think the main point of the ham mode is to allow more sending with more dBi / Watts.
At least that is what EU_433 HAM users tell me.
But you are right about convoluted and non-synched configs.
Like EU_868 is not a HAM band at all, only ISM, similar to CB ... yet nothing keeps the users for
- disabling duty cycle limit
- operating with higher dBi
when running with EU_868. That aspect is discussed here: https://github.com/meshtastic/firmware/discussions/6378#discussioncomment-13174571
Plus, then there is also a point of "HAM users should not be able to talk to non-HAM users on the same frequency",
something realistically possible at EU_433 in the Apps currently. 
See two comments with solutions: https://github.com/meshtastic/Meshtastic-Android/issues/1808#issuecomment-2864341847

shalberd, 2025-05-27:
This ticket could be closed if, by design, nodes connected to the app using HAM mode would not show non-HAM mode nodes in the app, or vice versa, non-HAM modes node when connected to the App would not show HAM mode users (is_licensed).

closed 2025-05-28 by jamesarich