meshtastic/Meshtastic-Android #1825: [Bug]: PR1051 overwrites default channel on import

akhepcat, 2025-05-07:
### Contact Details

discord: ak_hepcat

### What happened?

While the QR code feature provided by PR1051  is great, when *importing* a single channel  (exported on Android, imported on IOS)  the new channel wasn't *appended*  but instead overwrote the default channel.

This seems to be unintended behaviour - or at least something that should be optional on the import side.   The default on imports should be to *append* new channels, unless some new option "overwrite all channels"  is selected.

### App Version

2.5.16 (Android)

### Phone

Pixel 7,  Android 15

### Device

T-Deck 

### Firmware

2.6.6.54c1423

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

kaunix, 2025-05-07:
I imported his QR code in my meshtastic App on IOS. it overwrote my default channel deleted the other channels I had built.

garthvh, 2025-05-07:
The he default is to overwrite all channels, does the url for your QR code have ?add=true in it?

garthvh, 2025-05-07:
> > The default is to overwrite all channels, does the url for your QR code have ?add=true in it?
> 
> Inspecting the QR code url, it doesn't have any additional parameters (like ?add=) just the /e/#[code]
> 
> However, in further testing, the _Android_ app seems to handle this fine, but the _IOS_ app that [@kaunix](https://github.com/kaunix) is using doesn't.
> 
> I can close this here and open it up in the IOS app instead, since it doesn't seem Android related now.

Without the query string there is nothing to tell the client it should be addded.

closed 2025-05-07 by akhepcat