meshtastic/Meshtastic-Android #1838: [Bug]: Can't change LOCAL_ONLY in rebroadcast mode

Grim1313, 2025-05-11:
### Category

Other

### Hardware

T-Beam

### Firmware Version

2.6.6, 2.6.7

### Description

I can't change the rebroadcast mode setting. Whatever I choose, after saving, the setting remains in LOCAL_ONLY.
Lilygo T-beam SX1262, Android app 2.5.16.
FW: 2.6.6, 2.6.7

### Relevant log output

```Shell

```

madeofstown, 2025-05-14:
Pretty sure this is an app version issue.  When using alpha firmware make sure you are using the latest app build from GitHub, not the one in the Play Store.

Grim1313, 2025-05-14:
> Pretty sure this is an app version issue. When using alpha firmware make sure you are using the latest app build from GitHub, not the one in the Play Store.

I joined the beta testing program on Google Play and got version 2.5.25, which, as far as I understand, is the latest. However, still no luck.

jamesarich, 2025-05-14:
is this device (SX1262) specific? @Grim1313 have you tried with any other nodes/device types?

Grim1313, 2025-05-15:
@jamesarich No, unfortunately I only own one device, [this](https://lilygo.cc/products/t-beam-softrf?variant=43170139701429).

jamesarich, 2025-05-15:
> [@jamesarich](https://github.com/jamesarich) No, unfortunately I only own one device, [this](https://lilygo.cc/products/t-beam-softrf?variant=43170139701429).

I don't have that one - but i'll test on a few devices to see if it's device specific

Grim1313, 2025-05-20:
@jamesarich So, some new info here, it looks like this bug (?) only appears if the Licensed amateur radio slider is enabled.

jamesarich, 2025-05-20:
that leads me to believe it's a firmware setting from Licensed mode preventing this - @thebentern or @jp-bennett might be able to tag in here

garthvh, 2025-05-20:
I think this is likely, local only needs keys to decode packets, and there are no keys in ham mode

closed 2025-05-20 by garthvh