meshtastic/Meshtastic-Android #1864: [Bug]: App crashes when opening details about unrecognized device

EmilioAL-Git, 2025-05-17:
### Contact Details

emilioalfarolizan@gmail.com

### What happened?

I have an Elecrow Panel with MUI installed. Its suported but has no name yet inside the app so it shows as "unrecognized" 

When I tap to view details from other device the app crashes.

Here are some pictures explaining:

[https://i.ibb.co/fYCsk8qz/Screen-Recording-20250517-131208-Meshtastic.jpg](url)

[https://i.ibb.co/8LvTXBQj/Screen-Recording-20250517-131208-Meshtastic-1.jpg](url)

### App Version

2.5.16

### Phone

Samsung S24 Ultra

### Device

T1000e

### Firmware

2.6.4

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

jamesarich, 2025-05-17:
This has been resolved in newer alpha versions - and will make it to prod eventually. Feel free to join the 
google play test channel.

closed 2025-05-17 by jamesarich

EmilioAL-Git, 2025-05-17:
Thanks for quick reply!!