meshtastic/Meshtastic-Android #1869: [Bug]: Dark Mode in 2.6.0 very washed out

TheBamse, 2025-05-17:
### What happened?

Using dark mode in the new 2.6.0 app is quite washed out making it very hard to read.

[image]

### App Version

2.6.0

### Phone

Google Pixel 7, 14 (AP2A.240905.003)

### Device

RAK4631, T114v2

### Firmware

2.6.8, 2.6.7, 2.6.5

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

zomgugoff, 2025-05-17:
There is also a faint green color over everything

makanimike, 2025-05-18:
OP's screenshot is actually even pretty good. On my Pixel 8 it is entirely unreadable, even if I play around with the brightness setting.

tomdelta, 2025-05-19:
It's the same here. Dark mode ist not usable. [image]

PR #1872 (2025-05-19): fix: M3 bug squashing
- adressed theme issues with system bars
- updated messages to use secondary and tertiary theme colors, fixes #1869  
- Some rework on Settings/Connections screen ui and logic for Connection selection, Analytics and BugTracking controls.
-
closed 2025-05-19 by jamesarich