meshtastic/Meshtastic-Android #1878: [Bug]: v2.5.16 - Map last seen has an offset.

DaneEvans, 2025-05-20:
### Contact Details

github DaneEvans

### What happened?

In the lastest play store version (2.5.16) - Needs to be checked on the current head, but I'll do that over the next week or so. 
I've looked through the changelogs and not seen it appear there 

There's an offset on the last seen time on the Map. 
Last seen on Nodes list is accurate. 

Yesterday I drove past a few nodes at about midday. In the evening, I had a look at the app again, and noticed that the Map page was saying 24h, while the list was saying ~12.  This is still occuring 12 hours later, with 36h and 24h respectively. 

I'm in GMT+10, and went past the nodes between 10am and midday. So either of those could be source of the offset. 

### App Version

2.5.16

### Phone

Pixel 6a, Android 15. 

### Device

tracker-t1000e

### Firmware

2.6.4

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

b8b8, 2025-05-20:
Last seen on the map is when the last position packet was sent and where it was 

Last seen on the node list is the last of any packet.

DaneEvans, 2025-05-21:
Looks like that is the case, I can't see how it would be a valid position packet, but that's probably down to the node rather than the app. 
Closing

closed 2025-05-21 by DaneEvans