meshtastic/Meshtastic-Android #1895: [Bug]: More Infomration and Remote Admin no longer available

bquinlan, 2025-05-21:
### Contact Details

rdq@aslanrefuge.org

### What happened?

The menu that pops up when I tap on a node icon in the node view no longer includes the "More Information" option.  Among other things, that means there is no longer a way to access the Remote Admin feature.

### App Version

2.6.1

### Phone

Samsung S23 Ultra running Android 15

### Device

Heltec T114 

### Firmware

2.5.20.4c97351

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

b8b8, 2025-05-21:
Update your firmware.

bquinlan, 2025-05-21:
Sorry, I had the wrong version.  The firmware is 2.6.4.b89355f Beta.  I'm used to getting that from the More Information screen in the app.  :-}

I will try updating the 2.6.8 to see if it helps.

b8b8, 2025-05-21:
Yea please note if still an issue. I had issues with firmware prior to 2.6.5. 2.6.8 seems to work with no problems

jamesarich, 2025-05-21:
fixed in #1894

closed 2025-05-21 by jamesarich