meshtastic/Meshtastic-Android #1919: [Feature Request]: PAX counter graphical logging in the app.

Miaoucat0, 2025-05-23:
### Contact Details

Miaoucat

### Tell us your idea.

 i just updated to the lasted android release (from github) and i am so happy that pax module is now working!! (previously the received info didnt show on the app) there are so many uses for this i am having great plans for the future.

the thing is it would be great to have **data logging possible on other nodes thru the mesh?** because right now there is no notification, nor anything. the **PAX count** just show on the node list and thats it. it would be useful to have logging like we have for signal strengh and such. for example me i would want to know when the area is busy or not. since currently there is no "memory" i cannot see previous readings.  i know that graphical logging is already implemented in the app for battery levels and signal strength. it would be interesting to see it for PaxCounter too.

also on a related note some fringe uses might benefit from having a notification at "X" number PAX count, but i know it would be alot of programing to add it in. logging is the main thing needed right now.

example: 

[image]

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Miaoucat0, 2025-05-23:
@jamesarich Yes it is it. i dont know if the data is stored, but as of now it is not available in graph format. the received count is only shown in the node list.

[image]

Robert-0410, 2025-05-24:
> [@Robert-0410](https://github.com/Robert-0410) this sounds like your cup of tea
> 
> If I'm understanding [@Miaoucat0](https://github.com/Miaoucat0) 's request correctly, if we don't currently store/log pax packets - we could. From there we could display them in a graph and log like our other sensors.

Seems interesting, i'll keep it on my radar.

NumbarZero, 2025-06-23:
I thought it was implemented in the recent version but it isnt. PaxCounter cannot be of used as-is right now.

jamesarich, 2025-06-23:
> I thought it was implemented in the recent version but it isnt. PaxCounter cannot be of used as-is right now.

Even though the graph and log may not be implemented, Pax Counter stats are _still_ available on the node list cards - they haven't gone anywhere.

PR #2477 (2025-07-21): Feat/1919 pax graphs
depends on #2472 

closes #1919 

Adds PAX graphs. 
[image]

It may be time for us to hide metrics with no data, rather than grey them out.
closed 2025-07-21 by jamesarich