meshtastic/Meshtastic-Android #1926: [Bug]: Cannot easily switch nodes if previous node is out of range

jeremiah-k, 2025-05-24:
### Contact Details

Jack! on Discord

### What happened?

I meant to report this one earlier, I think it's been happening since v2.6.

When connected to a node and I move out of range of it and try to switch to another node, it is quite difficult to get it to connect to another node.

If I tap another node fast enough I'm able to get it to switch.

If I'm still in range of my current node when I try to switch, it works fine. 

### App Version

2.6.3

### Phone

Pixel 6 Pro/GrapheneOS 

### Device

RAK/T1000-E, Tbeam

### Firmware

2.6.8

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

b8b8, 2025-05-24:
Didn you try clicking "none" first then trying the second node?

jeremiah-k, 2025-05-24:
I'm not able to select None either. At least it won't refresh when I select none. 

I was always noticing this when I was out and about so I hadn't gone too far into testing. I wanted to report it before I forgot again. I just sat down and tried it some more. 

I think I see what's happening. If i select a node that's out of range it won't refresh the window until I select node that is in range.  Then it will redraw the page/move the button. 

There is a delay when selecting that, so I thought it wasn't working when I pressed it.

jeremiah-k, 2025-06-11:
@jamesarich when you have a few free moments, would you mind taking a look at this?  I might have a fix for this issue.
https://github.com/jeremiah-k/Meshtastic-Android/pull/1

I've been testing it on my Pixel 6 Pro (GrapheneOS) and it seems much smoother.