meshtastic/Meshtastic-Android #1933: [Bug]:

GoProProfi, 2025-05-25:
### What happened?

App notifies for every Node Clnnection, even for previously seen Nodes. Further more ther is no way go seperatly manage Notification settig s for the App.

### App Version

2.5.16

### Phone

Galaxy S24+, Android 15, One-UI 7.0

### Device

Lilygo T-Echo /

### Firmware

2.6.4.b89355f Beta

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

GoProProfi, 2025-05-25:
Thanks for your quick respons. As i understand this doesen't realy help me.
In sumary. I can't change the notification behaviour, unless i turn off al notification or have an non Samsund device. Correct?

So this issue sholud rather be an feature request to ad in appsettings to manage notification settings.

b8b8, 2025-05-25:
it is possible on samsung devices. you dont need to turn off all notifications.

GoProProfi, 2025-05-26:
Thanks this helped a lot. 

But the behafiour is saw was in my eyes worng. The app keept notifing me for the same Nodes it reached. I find this very anoying and would think of this feature more in kind of notifing me if ther was a connection to a new Node (one witch has never been seen).

b8b8, 2025-05-26:
You are also running the app from January. I suggest updating to 2.6.3 in the beta channel. A lot has changed in 5 months.

GoProProfi, 2025-05-26:
I am now running 2.6.2 but saw the same issue.

closed 2025-06-22 by jamesarich