meshtastic/Meshtastic-Android #1954: [Feature Request]: Support for telemetry properties to support soil moisture & temp

Justin-Mann, 2025-05-27:
### Contact Details

discord: mcluvin7930

### Tell us your idea.

I'd like the app to...
support and display soil moisture sensor info.  there is a firmware pr () around adding support for the RAK12035 soil moisture sensor, this ask is to add support for the new protobuff properties added to the environmental telemetry and add display elements that display similar to the other environmental metrics, logging.. any other elements I might not be thinking about atm.

the new properties are
...
    /* Soil moisture measured (% 1-100) */
    bool has_soil_moisture;
    uint8_t soil_moisture;
    /* Soil temperature measured (*C) */
    bool has_soil_temperature;
    float soil_temperature;
} meshtastic_EnvironmentMetrics;
...

and should be handled similarly to
...
    /* Temperature measured */
    bool has_temperature;
    float temperature;
    /* Relative humidity percent measured */
    bool has_relative_humidity;
    float relative_humidity;
...

### Relevant log output

```shell
Processing rak4631 (board: wiscore_rak4631; platform:
platformio/nordicnrf52@^10.8.0; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/wiscore_rak4631.html
PLATFORM: Nordic nRF52 (10.8.0) > WisCore RAK4631 Board
...
Building .pio\build\rak4631\firmware.zip
Zip created at .pio\build\rak4631\firmware.zip
...
//\ E S H T /\ S T / C

DEBUG | ??:??:?? 4 Filesystem files:
DEBUG | ??:??:?? 4  adafruit (directory)
DEBUG | ??:??:?? 4  bond_prph (directory)
DEBUG | ??:??:?? 4  bond_cntr (directory)
DEBUG | ??:??:?? 4    MODE (1 Bytes)
DEBUG | ??:??:?? 4    RAK (108 Bytes)
DEBUG | ??:??:?? 4    meshtastic.txt (54 Bytes)
DEBUG | ??:??:?? 4  prefs (directory)
DEBUG | ??:??:?? 4    module.proto (120 Bytes)
DEBUG | ??:??:?? 4    channels.proto (155 Bytes)
DEBUG | ??:??:?? 4    device.proto (122 Bytes)
DEBUG | ??:??:?? 4    config.proto (204 Bytes)
DEBUG | ??:??:?? 4    nodes.proto (1096 Bytes)
DEBUG | ??:??:?? 4 Power::lipoInit lipo sensor is not ready yet
DEBUG | ??:??:?? 4 Use analog input 5 for battery level
INFO  | ??:??:?? 4 Scan for i2c devices
DEBUG | ??:??:?? 4 Scan for I2C devices on port 1
INFO  | ??:??:?? 4 RAK12035 found at address 0x20
DEBUG | ??:??:?? 4 Register MFG_UID FE: 0xffff
DEBUG | ??:??:?? 4 Register MFG_UID 05: 0xffff
INFO  | ??:??:?? 4 1 I2C devices found
...
INFO  | ??:??:?? 15 [EnvironmentTelemetry] Environment Telemetry: init
INFO  | ??:??:?? 15 [EnvironmentTelemetry] Init sensor: RAK12035
INFO  | ??:??:?? 19 [EnvironmentTelemetry] Sensor1 Firmware version: 2
INFO  | ??:??:?? 19 [EnvironmentTelemetry] RAK12035Sensor Init Succeed
INFO  | ??:??:?? 19 [EnvironmentTelemetry] Opened RAK12035 sensor on i2c bus
INFO  | ??:??:?? 21 [EnvironmentTelemetry] Dry calibration value is 558
INFO  | ??:??:?? 21 [EnvironmentTelemetry] Wet calibration value is 395
INFO  | ??:??:?? 21 [PowerTelemetry] Power Telemetry: init
...
DEBUG | ??:??:?? 85 [Power] Battery: usbPower=1, isCharging=1, batMv=4216, batPct=100
INFO  | ??:??:?? 98 [EnvironmentTelemetry] Send: barometric_pressure=0.000000, current=0.000000, gas_resistance=0.000000, relative_humidity=0.000000, temperature=0.000000
INFO  | ??:??:?? 98 [EnvironmentTelemetry] Send: voltage=0.000000, IAQ=0, distance=0.000000, lux=0.000000
INFO  | ??:??:?? 98 [EnvironmentTelemetry] Send: wind speed=0.000000m/s, direction=0 degrees, weight=0.000000kg
INFO  | ??:??:?? 98 [EnvironmentTelemetry] Send: radiation=0.000000##R/h
INFO  | ??:??:?? 98 [EnvironmentTelemetry] Send: soil_temperature=25.000000, soil_moisture=3
DEBUG | ??:??:?? 98 [EnvironmentTelemetry] Partially randomized packet id 2199595487
INFO  | ??:??:?? 98 [EnvironmentTelemetry] Send packet to mesh
...
```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Miaoucat0, 2025-06-03:
i dont know anything about the RAK module, but i am extremely interested in Soil Moisture sensors. i would even buy one of their sensor if its the only solution available on Meshtastic. +1

Justin-Mann, 2025-06-18:
@Miaoucat0 ... for now... https://www.instructables.com/RAKMeshtastic-Based-Irrigation-Monitor/

Is RAK4361 based and target a RAK monitor RAK12035, but I am using this IRL.

Working on getting this into the main firmware branch as I have time.. I really only put this feature request here so I can pick it up later.

closed 2025-07-22 by DaneEvans