meshtastic/Meshtastic-Android #1980: [Feature Request]: Traceroute Cooldown UI

jamesarich, 2025-05-29:
### Contact Details

@olm3c on discord

### Tell us your idea.

Traceroutes have an enforced 30s firmware cooldown. As a user I'd like to have that also visually represented in the client so I know when I can send another traceroute.

In the ui components where the Traceroute action controls exist (Node chip dropdown menu, Node details) the component should be disabled (`enabled=false`) and a 30 second progress indicator should be shown in-line with, below, or as part of the action control.

This functionality was recently introduced on iOS and would be an excellent and easy add to the Android app.

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

macvenez, 2025-06-01:
I'm using a firmware which has been modified a little bit and the default traceroute interval is 60 seconds, would it be possible to read the interval set from the device or leave it as it is?

garthvh, 2025-06-01:
> I'm using a firmware which has been modified a little bit and the traceroute interval is 60 seconds, would it be possible to read the interval set from the device or leave the app behaviour as it is?

This is not a value that can be read from the api.

macvenez, 2025-06-01:
Actually I believe the interval could be read by sending two consecutive requests and reading the notification which is sent by the device showing the timeout interval. (At least this is how it works on Android App).

That won't be useful if also setting a lower minimum traceroute interval than 30 seconds as you'll be limited by the app anyway (maybe on higher presets like ShortFast)

jamesarich, 2025-06-01:
completed in #1987

closed 2025-06-01 by jamesarich