meshtastic/Meshtastic-Android #2019: [Bug]: Clicking "more details" for a specific node causes app to crash

Rhysun, 2025-06-04:
### Contact Details

rhysun@gmail.com

### What happened?

I have deployed a solar node made with a XIAO nRF52840 and Wio-SX1262 kit. The node works as expected, successfully routing messages and all the other stuff. I set it up via the CLI with the admin passkey set so I could fine tune the config remotely. However, when I click the "more details" button for this node the app immediately crashes. The "more details" button works correctly for all other nodes on the node list.

All other nodes on the list specifies what type of device they are (rpi-pico, seeed-xiao-s3 etc.) but this node is shown as "unrecognised".

I've tried reinstalling the device's firmware, double checking that it is the correct one, and I've reinstalled the app on my phone (Samsung Galaxy S10e), but the same behaviour recurs.

### App Version

2.5.16

### Phone

Samsung Galaxy S10e

### Device

Seeed XIAO nRF52840 kit

### Firmware

2.6.4.b89355f Beta

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Vilda007, 2025-06-04:
I am experiencing the same behavior when requesting "more details" for a node running on Heltec MeshPocket (which is not recognized and reported by web client as "Unknown").

jamesarich, 2025-06-05:
this crash is caused by that app's hardware list beign out of date - this has been resolved in the latest alpha versions of the app

closed 2025-06-05 by jamesarich