meshtastic/Meshtastic-Android #2021: [Feature Request]: In location map, color the most recent point differently

alexholman, 2025-06-04:
### Contact Details

aholman@gmail.com

### Tell us your idea.

Currently all locations appear as the triangle connected by lines.
Have some call-out or indication of the most recent point. Either minimize the icon for the old points to a dot, or a smaller arrow, or color highlight the newest point.

**Use case:** I'm tracking a runner over multiple laps of a course. After several overlapping laps, it's hard to determine where on course they actually are at the last location message.

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

b8b8, 2025-07-12:
@jamesarich please reopen.

jamesarich, 2025-07-12:
fwiw @b8b8 this is in the google maps branch already

tteskac, 2025-07-23:
I actually implemented the other day a similar thing for the main map, so I might actually look into your request and propose a PR for it.

FYI: for the feature I implemented (PR not created yet but I might if other's find it helpful), I wanted to visualize most recent "heard" nodes, which gives immediate feedback what nodes I'm most likely able to communicate with at the current location. Color is Meshtastic primary fading into white for last heard time between 0-30min.

[image]