meshtastic/Meshtastic-Android #2039: [Bug]: Sending channel configuration update prevents sending another one until re-entering Channels page

rfschmid, 2025-06-05:
### Contact Details

russ@sumonkey.com

### What happened?

1. Radio Configuration > Channels
2. Observe "Send" and "Cancel" buttons are disabled because no changes have been made yet
3. Make some change, eg re-order channels, delete a channel, add a channel, change position precision for a channel etc.
4. Observe "Send" and "Cancel" are now enabled
5. Press "Send", wait for "Delivery confirmed" and close that dialog
6. Observe "Send" and "Cancel" are disabled again since there are no changes to send
7. Make another change as in step 3
8. Observe "Send" and "Cancel" buttons are still disabled, despite having changes to send

Must back out of Channels and re-enter the page to make additional changes. 

The "Send" and "Cancel" buttons should be re-enabled after step 7 so user can send additional changes to the radio without needing to back out of the channel config page and re-push it.

### App Version

2.6.9

### Phone

Pixel 7

### Device

Heltec T114

### Firmware

2.6.7

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

davs2rt, 2025-06-20:
Also seen in Android 2.6.19, but I still see the Cancel button enabled, Send is not enabled.

DaneEvans, 2025-07-07:
still present in 2.6.29

closed 2025-07-11 by DaneEvans