meshtastic/Meshtastic-Android #2077: [Bug]: blank secondary channel names are incorrect when editing channels

battlehax, 2025-06-11:
### What happened?

When going to Radio Configuration, then Channels, my channel name for LongFast, shows the name of my primary channel instead. When going to channels on the bottom bar, it shows correctly, but once I click edit, it's wrong again.

After some discussion, we determined that this doesn't happen when the channel is explicitly named "LongFast", only when the channel's name is blank. The proper way to set that up and name per platform is explained here <https://meshtastic.org/docs/configuration/tips/#creating-a-private-primary-with-default-secondary>

Further testing seems to reveal that sharing channel urls between devices and using the cli to backup/restore settings can blank the field.

[image]

[image]

### App Version

2.6.15

### Phone

Pixel 6a, Android 16

### Device

Heltec v3

### Firmware

2.6.5

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

battlehax, 2025-06-11:
> Is it an issue when restoring channels with the QR code?
> 
> Restoring channels with the config backup (in android) hasn't worked for a while, but restoring the longfast secondary channel has always worked (for me) using the QR.

It should be okay when generating the QR code on Android. I will try to investigate if sharing a QR from an iOS device with a blank channel name to an Android device could cause this.

garthvh, 2025-06-11:
Every newly flashed device has a channel with a blank channel name, LongFast is not the actual channel name, the hash for a channel falls back to the Modem Preset if a name is not set.