meshtastic/Meshtastic-Android #2094: [Bug]: Map Crash

Rikku2000, 2025-06-12:
### What happened?

A bug happened!

I found a bug in the Android version 
2.6.13

When you are switching multiple times from nodes to map, the map grayed out once again switching, the app will crash immediately. 

### App Version

2.6.13

### Phone

Android 11

### Device

Heltec v3

### Firmware

2.6.13

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

prokrypt, 2025-06-12:
I can reproduce it.

portions of logcat that may or may not be relevant:

```
06-12 14:15:14.561 19971 19971 E OsmDroid: Error initializing archive file provider /storage/6436-3339/Android/data/com.geeksville.mesh/files/osmdroid/mainFile.sqlite
06-12 14:15:14.561 19971 19971 E OsmDroid: java.lang.InstantiationException: java.lang.Class<coil3.util.DrawableUtils> cannot be instantiated
06-12 14:15:14.561 19971 19971 E OsmDroid:      at java.lang.Class.newInstance(Native Method)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at org.osmdroid.tileprovider.modules.MapTileFileArchiveProvider.findArchiveFiles(SourceFile:91)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at org.osmdroid.tileprovider.modules.MapTileFileArchiveProvider.<init>(SourceFile:36)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at org.osmdroid.tileprovider.MapTileProviderBasic.<init>(SourceFile:111)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at org.osmdroid.views.MapView.<init>(SourceFile:176)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at com.geeksville.mesh.ui.map.MapViewWithLifecycleKt.rememberMapViewWithLifecycle(SourceFile:38)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at com.geeksville.mesh.ui.map.MapViewWithLifecycleKt.rememberMapViewWithLifecycle(SourceFile:5)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at com.geeksville.mesh.ui.map.MapViewKt.MapView(SourceFile:471)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at com.geeksville.mesh.navigation.NavGraphKt$NavGraph$1$1$1.invoke(SourceFile:2)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at com.geeksville.mesh.navigation.NavGraphKt$NavGraph$1$1$1.invoke(SourceFile:1)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(SourceFile:28)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(SourceFile:3)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at androidx.compose.material3.ButtonKt$Button$2$1.invoke(SourceFile:57)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(SourceFile:11)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(SourceFile:1)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at androidx.compose.runtime.AnchoredGroupPath.CompositionLocalProvider(SourceFile:77)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at androidx.compose.runtime.saveable.SaveableStateHolderImpl.SaveableStateProvider(SourceFile:96)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at kotlin.jdk7.AutoCloseableKt.SaveableStateProvider(SourceFile:133)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at androidx.compose.material3.ButtonKt$Button$2$1.invoke(SourceFile:99)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(SourceFile:11)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(SourceFile:1)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at androidx.compose.runtime.AnchoredGroupPath.CompositionLocalProvider(SourceFile:29)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at kotlin.jdk7.AutoCloseableKt.LocalOwnersProvider(SourceFile:94)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at androidx.navigation.compose.NavHostKt$NavHost$32.invoke(SourceFile:123)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(SourceFile:28)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(SourceFile:3)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at androidx.compose.foundation.text.TextFieldCursorKt$cursor$1.invoke(SourceFile:151)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(SourceFile:20)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(SourceFile:2)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at androidx.compose.animation.Scale.AnimatedEnterExitImpl(SourceFile:1310)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at androidx.compose.animation.AnimatedContentKt$AnimatedContent$6$1.invoke(SourceFile:265)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(SourceFile:11)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(SourceFile:1)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at androidx.compose.animation.AnimatedContentKt.AnimatedContent(SourceFile:881)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at kotlin.text.CharsKt.NavHost(Unknown Source:354)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at androidx.navigation.compose.NavHostKt$$ExternalSyntheticLambda4.invoke(SourceFile:76)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at androidx.compose.runtime.ComposerImpl.recomposeToGroupEnd(SourceFile:548)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at androidx.compose.runtime.ComposerImpl.skipCurrentGroup(SourceFile:145)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at androidx.compose.runtime.ComposerImpl.doCompose-aFTiNEg(SourceFile:121)
06-12 14:15:14.561 19971 19971 E OsmDroid:      at androidx.compose.runtime.ComposerImpl.recompose-aFTiNEg$runtime_release(SourceFile:31)
[... cut ...]
06-12 14:15:14.979 19971 19971 E AndroidRuntime:        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:992)
06-12 14:15:14.979 19971 19971 E AndroidRuntime:        Suppressed: kotlinx.coroutines.internal.DiagnosticCoroutineContextException: [androidx.compose.ui.platform.AndroidUiFrameClock@163936a, androidx.compose.ui.platform.MotionDurationScaleImpl@c4c405b, StandaloneCoroutine{Cancelling}@97550f8, AndroidUiDispatcher@d6f39d1]
06-12 14:15:14.980  2814  5058 W ActivityTaskManager:   Force finishing activity com.geeksville.mesh/.MainActivity
```

b8b8, 2025-06-12:
What basemap were you using?

PR #2143 (2025-06-17): fix(#2094): remove redundant mapView.onDetach() call
Removes redundant onDetatch in favor of lifecycle handling.
Fixes NPE crash when navigating into and away from the map quickly.

fixes #2094
closed 2025-06-17 by jamesarich