meshtastic/Meshtastic-Android #2121: [Bug]: Battery Level Display Inconsistency

Phi-Wa, 2025-06-15:
### What happened?

At the most recent timestamp **6/14/25 20:15:09**, the battery level is shown as **97% with a voltage of 4.15V** in the data table below the graph. However, the corresponding point on the green battery level line (far right of the graph) appears to be closer to **80%,** not 97%.

This discrepancy suggests that the graph does not accurately reflect the actual battery percentage at that specific time.

[image]

**Steps to Reproduce**:
1. Navigate to Nodes → Device Metrics Log.
2. Select the relevant device.
3. Locate the timestamp (here: 6/14/25 20:15:09.)
4. Compare the battery percentage value in the table (97%) with the green graph line position.

**Expected Result**:
The green line in the graph should align with the 97% battery level reported in the table for the same timestamp.

**Actual Result**:
The green line at the specified time appears to be around 80%, indicating a visual or scaling mismatch between the graph and the tabular data.

### App Version

2.6.15

### Phone

Samsung Galaxy S21 Ultra
Android 15

### Device

Heltec V3.2

### Firmware

2.6.11.60ec05e Alpha

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

DaneEvans, 2025-06-15:
Have you scrolled all the way right? The top time stamp on the right is the whole interval, not just what's showing. 

#2100

Phi-Wa, 2025-06-19:
Thanks for the clarification. You're right.

I was misled by the fixed timestamp at the top and didn’t realize the graph continued further to the right. After scrolling all the way, I can see that the values are displayed correctly.

Since this kind of misinterpretation is already described in https://github.com/meshtastic/Meshtastic-Android/issues/2100 (... _When looking at device metrics & signal metrics at all timescales, the end x-axis values remain fixed even when scrolling the graph..._"), I think it’s fine to close this report as a duplicate or related UX issue.

Appreciate the quick response!

closed 2025-06-19 by jamesarich