meshtastic/Meshtastic-Android #2128: [Bug]: Unable to change Map Reporting Interval (MQTT settings)

Nevermore-42, 2025-06-15:
### Contact Details

Discord nevermore42_

### What happened?

The android client no longer allows changing the default value of the Map Reporting Interval under the MQTT settings. I've tried multiple entries, below and above the default 3600s, but everytime it displays error message as in the attached imaged (not allowing you to save changes)

[image]

### App Version

2.6.16

### Phone

Pixel 7, Android 15

### Device

Lilygo T-Echo

### Firmware

2.6.11.60ec05e Alfa

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

garthvh, 2025-06-15:
One hour is the correct lower limit.

skgsergio, 2025-06-15:
> One hour is the correct lower limit.

Thing is that if you try to set 2h, 12h (43200 as in the screenshot in the issue)... You just can't.

The app only allows 3600 as value.

skgsergio, 2025-06-16:
cc @jamesarich as it seems this commit is where the fetch and storage of the value was lost https://github.com/meshtastic/Meshtastic-Android/commit/3115bbe58daf0ebe104b3875b244f0da8944d32f#diff-d4e50a3c3218f66e33ac5cd4820bde36532276da0ffec200636ac0c76f6e7e3a

closed 2025-06-16 by jamesarich