meshtastic/Meshtastic-Android #2162: [Bug]: Add *slight* bias to message list

b8b8, 2025-06-19:
### What happened?

The newest update allows far more room for text and eliminates the data wrapping issue of the previous release by moving the emoji icon and the nodeID tablet. This can make it slightly hard to sort through sent vs received messages (yes i know your own message doesnt have a nodeID tablet). 

I think there should be a slight right side and left side bias to differentiate messages.

See below.

Before
[image]

After
[image]

### App Version

2.6.20

### Phone

Pixel 9a

### Device

N/A

### Firmware

2.6.12

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

closed 2025-06-22 by jamesarich

prokrypt, 2025-06-23:
fix'd in 2.6.24!

[image]

jamesarich, 2025-06-23:
> fix'd in 2.6.24!

nice amongus
[image]

prokrypt, 2025-06-23:
> > fix'd in 2.6.24!
> 
> nice amongus [image]

someone should make it into an on-device emoji

jamesarich, 2025-06-23:
[image]
 quickchat tho ඞ

prokrypt, 2025-06-23:
> quickchat tho ඞ

[image]
Keyboard dictionary tho ඞ

jamesarich, 2025-06-23:
> > quickchat tho ඞ
> 
> [image] Keyboard dictionary tho ඞ

kinda sus ඞ

prokrypt, 2025-06-23:
> > > quickchat tho ඞ
> > 
> > 
> > [image] Keyboard dictionary tho ඞ
> 
> kinda sus ඞ

[image]