meshtastic/Meshtastic-Android #2213: [Feature Request]: Debug panel filters mode

lkosson, 2025-06-22:
### Tell us your idea.

Related to #2167. Currently all active filters are `OR`ed together. It makes sense for filtering based on same property (e.g. port name), but isn't too useful when filtering by different properties (e.g. port name and sender).

I would like to either:
 * be able to switch manually between `OR`ing or `AND`ing active filters
 * have app automatically `OR` filters related to same property (port, node, date etc) and `AND` all those `OR`ed groups

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

lkosson, 2025-06-22:
Switching between all `AND`s or all `OR`s is a good start and suits most of my needs. Only reasonable use case for `OR`ing and `AND`ing I can think of is providing multiple node IDs or multiple ports. Both of those type of values can be easily detected by looking at filter text.

In other words:
 * by default `AND` all the filters, i.e. require all clauses to be present in packet's `toString` to show it
 * if there are multiple filters matching pattern `![0-9A-Fa-f]{8}`, require only one of them to be present in packet's `toString`
 * if there are multiple filters named exactly like a predefined port name, require only one of them

That way you don't need to parse actual packet data or filter expressions and still do matching by comparing strings instead of actual properties.

For anything more complex, I'd rather browse packet log on PC anyway.

DaneEvans, 2025-06-27:
#2265 implements AND

See how that goes for a few weeks, and see if we need anything more. 
The dump to a file will allow further analysis on a pc if needed.

closed 2025-07-03 by DaneEvans