meshtastic/Meshtastic-Android #2214: [Bug]: Debug Menu UI Bugs

jamesarich, 2025-06-22:
### Contact Details

olm3c on discord

### What happened?

related to #2167 

[image]

The logs are visible below the search and filter views and look a bit wonky, maybe consider an opaque or mostly opaque background for those views.

Bonus points for animated visibility/hiding while scrolling to maximize viewing space.

### App Version

current main / 2.6.22

### Phone

-

### Device

-

### Firmware

-

### Relevant log output

```shell
-
```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

DaneEvans, 2025-06-22:
I'll move these to here then:  

- [ ] Landscape view on small HW is not well thought out - will likely need a completely different location for active filters.
     - [ ] Also look at the search bar and filter text and how they blow out.

jamesarich, 2025-06-22:
closed by #2219

closed 2025-06-22 by jamesarich