meshtastic/Meshtastic-Android #2216: [Bug]: unread message count not updating

lkosson, 2025-06-22:
### What happened?

When the message list contains more than one page of read messages and new (unread) message is received, opening the conversation for the first time does not reset unread status of the message (even though it scrolls down to it) and does not update unread message count. Opening the conversation once again properly marks message as read.

[image]

### App Version

2.6.11

### Phone

Emulator, Pixel 6a

### Device

(unrelated)

### Firmware

(unrelated)

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

lkosson, 2025-06-23:
With some recent changes, unread message count stopped updating altogether. Might has something to do with `MessageList.UpdateUnreadCount` using `firstVisibleItemIndex` and first unread message not being at or above first visible message.

[image]

brebtatv, 2025-06-23:
The Android app 2.6.4 problem is back and even worse. There is no way around it.