meshtastic/Meshtastic-Android #2250: [Bug]: Opening a contact share link does not add the contact to the node list.

StukyX, 2025-06-25:
### What happened?

My friend tried to share their Meshtastic contact with me using the generated share link (`https://meshtastic.org/v/#...`). When I tapped the link, it first opened a page in my web browser (the page was blank not sure if it is intentional). On that page, I tapped the browser button to "Open with Meshtastic." The Meshtastic app launched as expected, but then nothing happened. The app just opened to its main screen, and my friend's device did not appear in my node list.

When I scanned the QR code directly within the app, it worked perfectly and added his device to my node list. 

### App Version

2.6.24

### Phone

POCO X5 Pro 5G, Android 14

### Device

Heltec V3

### Firmware

2.6.11

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

danielb7390, 2025-06-27:
Having the same problem.
Where do you scan the node from within the app? I only found the scan for the channel settings

jamesarich, 2025-07-31:
i get all notifications for issues - a mention isn't super necessary

PR #2336 (2025-07-03): feat: Add support for sharing contacts via deeplink
This commit introduces the ability to share contacts using QR codes or shareable links.

Key changes:
- Added `sharedContactRequested` StateFlow to `UIState.kt` to manage contact import requests.
- Implemented logic in `MainActivity.kt` to handle app links for shared contacts (URLs starting with "/v/" or "/V/").
- Updated `ContactSharing.kt` to use `sharedContactRequested` for managing the contact import dialog.
- The app now presents a confirmation dialog when a shared contact is received via QR code scan **or** link.

resolves #2250
closed 2025-09-08 by github-actions[bot]