meshtastic/Meshtastic-Android #2259: [Bug]: Accessibility: a couple of issues

shortwavesurfer2009, 2025-06-26:
### What happened?

1. On the nodes screen, the node chip at the top right hand corner just before the overflow menu button reads as unlabeled.
2. It is no longer possible for a talkback user to get to the navigation bar to access messages, nodes, channels, etc.
3. Elements are not being read top to bottom properly on the chat screen. See video included below.

https://github.com/user-attachments/assets/efa47dc2-4ead-410a-89d2-2e1c56a0143f

### App Version

2.6.24

### Phone

OnePlus Nord N200

### Device

Seeed tracker t1000e

### Firmware

2.6.12

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

DaneEvans, 2025-08-05:
Just remove the stale tag and reopen it.

DaneEvans, 2025-09-07:
After the refactor, the bottom bar is still not included in talkback. 
Haven't looked at the ordering, as the emulator doesn't let you do a 2 finger swipe

shortwavesurfer2009, 2025-09-07:
The two finger swipes is only if you were on a physical device. If talkback is already on, you don't need to do any of the two finger gestures.

DaneEvans, 2025-09-07:
I literally can't scroll through them on an emulator, either help, or leave it to us.
And no, creating issues complaining about accessibility across dozens of repos is not *helping*

DaneEvans, 2025-09-07:
I invite you to read your own git history. And past warnings. You may find it enlightening.  

In the meantime, I've blocked you and won't see your tickets again. 

Good luck finding a Dev to pick it up.