meshtastic/Meshtastic-Android #2286: [Feature Request]: Improve input fields for telemetry intervals

Mictronics, 2025-06-28:
### Tell us your idea.

The input fields for telemetry intervals are marked to be in seconds, which is true for the entered value but they are indicating only the users expectation not what happens in firmware.
With the actual calculation method in firmware the telemetry interval depends on (in addition to user value) mesh size (number of online nodes) and modem preset.

Example calculation for a mesh with 100 nodes online on Long_Fast, the telemetry interval is set to 3600 seconds: `(((100 - 40) * 0.075) + 1) * 3600 = 19800 seconds or 5.5 hours` 
100 = online nodes
40 = below 40 online nodes only a factor is taken into account
0.075 = Preset factor for Long_Fast
3600 = user selected interval value

Under such condition the real telemetry interval is significantly different from what the user expects by his selected values. Awareness should be raised or the firmware calculation taken into account to indicate interval values to the user that fit more the reality.

I was already looking for problems because of that, since I closely monitor my nodes outside in the wild, when telemetry was not reported within the expected period.

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

b8b8, 2025-06-28:
A note should be added to the telemetry config page that intervals are automatically scaled in meshses larger than 40 nodes

"Telemetry Config (Auto-scaled when online nodes >40)"

Mictronics, 2025-06-28:
The idea was not to update the number in settings depending on dynamic online node count.
But a bit more explanation is needed, IMHO.

b8b8, 2025-06-28:
> or the firmware calculation taken into account to indicate interval values to the user that fit more the reality

closed 2025-08-05 by github-actions[bot]