meshtastic/Meshtastic-Android #2333: [Feature Request]: Streamline actions

DaneEvans, 2025-07-03:
### Contact Details

@DaneEvans

### Tell us your idea.

I'm seeing some delays in actions, 
My guess is that multiple workflows are being triggered on MRs. 
Particularly when they are being updated to main, 

I'm going to have a look and see if we can improve it. 
- At the very least detekt can probbaly be skipped on merge queue. 
- Build can maybe be skipped on MRs (Test will cover that) 
- Maybe only a single test on MRs, both on Main / merge queues

Make sure that we don't have exact duplicates as well. 

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

DaneEvans, 2025-07-03:
#2322  May have fixed it, but I'll take a look and see if I can spot anything

jamesarich, 2025-07-03:
yep - i did a bit of tuning today, should be _better_ but i would imagine there's still room to deduplicate some of these checks

DaneEvans, 2025-07-06:
- Do all the cronjob ones need to be hourly, 2-3 would seem adequate. 
- make sure that they stop when a newer commit happens

DaneEvans, 2025-07-07:
#2361

closed 2025-07-07 by DaneEvans