meshtastic/Meshtastic-Android #2335: [Feature Request]: Auto enable client_mute >=10,000ft

shortwavesurfer2009, 2025-07-03:
### Tell us your idea.

I see people on flights who have their nodes in client mode which causes connections to nodes miles away that you will never be able to hear from again.

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

jamesarich, 2025-07-03:
how tho

shortwavesurfer2009, 2025-07-03:
To add to this a little bit, if the node tells the app that it's above 10,000 feet, this could occur. Otherwise, the app could see if it's above 10,000 feet using the phone's GPS. And if neither are available, it would do nothing at all.

You would want the app to be the one to control this instead of the firmware because otherwise you couldn't launch any kind of balloons with mesh gear as some amateur radio clubs do.

DaneEvans, 2025-07-05:
- 10000 feet is low in many parts of the world unless we went AGL, and then we need a way to get GL data 
- how often would we even have phone or node GPS? 
- GPS elevation data is notoriously noisy

shortwavesurfer2009, 2025-07-05:
With GPS, you ought to be able to calculate movement speed across the planet. So theoretically, you could set it to if the device is over 300 miles per hour instead of a height.

As far as how often GPS is on during a flight either by the phone or by the node, at least in personal experience, I've not noticed any without it on, but that doesn't mean that it doesn't happen.

DaneEvans, 2025-07-05:
You could. is there any value? I'm not convinced.

Not enough to add 5 extra permissions, handle interfacing and fall throughs for multiple different mechanisms, for something that is not even a bugbear

shortwavesurfer2009, 2025-07-05:
200 miles per hour is 293 feet per second. So you could assume a default speed of zero and measure for a couple of seconds and take the average. And if the average was equal to or greater than 293 feet of acceleration, you could auto enable client mute.

closed 2025-07-05 by jamesarich