meshtastic/Meshtastic-Android #2394: [Feature Request]: Tooltips for channel indicators

battlehax, 2025-07-08:
### Tell us your idea.

It would be nice, especially with the new indicators added in https://github.com/meshtastic/Meshtastic-Android/pull/2356, to have some way to get a little more detail about the meaning of the icon. A tooltip when tapping the icon in the header of a channel would be very helpful!

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

miketweaver, 2025-07-08:
It would be awesome if the tool tip indicated which triggers caused the status, rather than a general tooltip. I believe multiple things can cause the icon to change types and indicating what is causing it would be helpful!

jamesarich, 2025-07-08:
#2334 has an example of the ios info popup - maybe could be replicated here

closed 2025-07-20 by jamesarich