meshtastic/Meshtastic-Android #2408: [Bug]: inconsistent use of Node Id vs User Id in UI

oseiler2, 2025-07-10:
### What happened?

By default a node's owner.id (aka user Id) is generated by using a '!' followed by the hex representation of its node id.
In some cases a node might have a different owner.id. While this cannot easily be set by a user (afaik), it can be a result of a node collision, as we have seen happening in the wild.

When clicking on a node in the node list to look at its details the app displays 'User ID' followed by the hex representation of the nodeId, but not the actual owner.id. This should be the actual owner.id and not derived from the nodeId.
[image]

In the User Config screen of the node configuration the app displays 'Node ID' followed by the owner.id string. This should be labelled User ID and not Node ID (or, if the Node ID is meant to be shown here, the label is wrong and there shoudln't be the '!')
[image]

### App Version

2.6.30

### Phone

Pixel 7a

### Device

any

### Firmware

2.5.20, 2.6.11, 2.7.1

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

erayd, 2025-07-10:
I haven't been able to replicate this on v2.6.29 of the Android app, but can confirm that the bug is present on v2.6.10 of the iOS app. That node has an ID of `!mesh_tips`, and a node number of `4`.

[image]

garthvh, 2025-07-10:
> I haven't been able to replicate this on v2.6.29 of the Android app, but can confirm that the bug is present on v2.6.10 of the iOS app. That node has an ID of `!mesh_tips`, and a node number of `4`.
> 
> <img alt="Image" width="273" height="345" src="https://private-user-images.githubusercontent.com/121486/464629646-9589fca1-34e5-479f-a24f-cbd191f75e22.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTIxNjkzMzksIm5iZiI6MTc1MjE2OTAzOSwicGF0aCI6Ii8xMjE0ODYvNDY0NjI5NjQ2LTk1ODlmY2ExLTM0ZTUtNDc5Zi1hMjRmLWNiZDE5MWY3NWUyMi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwNzEwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDcxMFQxNzM3MTlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xNzliYTBmMTRlYzMxYmUxZDY5OTUxNjE2OGRhN2E0ZWU5Mzc5NTczZWVlODQxYzRiNmUxMmRkMmZiMmMxMDcyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.9O_Q773QIenb7UayCAQfAlg1UXbfc_GgLNVrrkNEk7I">

Hex of 4 is 4

garthvh, 2025-07-10:
> > About a month ago we added coercion of user.id to the firmware
> 
> Fair enough, so I assume the intention is to retire user.id?
> 
> Please note that the User Config screen still shows the actual user.id (using the label 'Node ID', not User ID) https://github.com/meshtastic/Meshtastic-Android/blob/main/app/src/main/java/com/geeksville/mesh/ui/radioconfig/components/UserConfigItemList.kt#L96
> 
> <img alt="Image" width="404" height="95" src="https://private-user-images.githubusercontent.com/3577835/464938600-435f3f7c-a44e-49e3-b3b8-704b2bff8763.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTIxODM2MzUsIm5iZiI6MTc1MjE4MzMzNSwicGF0aCI6Ii8zNTc3ODM1LzQ2NDkzODYwMC00MzVmM2Y3Yy1hNDRlLTQ5ZTMtYjNiOC03MDRiMmJmZjg3NjMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDcxMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA3MTBUMjEzNTM1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZWQ5ZmU4MjM5YWVjMmQ1OTUxNDMxYmZmMDkwOWY2YzU3OTgzYmI2MjlkMTMzZjNmNDM5MDZiMmQyZjlmZmY3YSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.-Q6Xz-k-aGx31DG2fWIfpnpsezbzfKGAoltEq0UOV6Q">

It should always generated from node number by clients.

oseiler2, 2025-07-10:
> It should always generated from node number by clients.

Sure. But it isn't in case of a node-id collision as shown above.

garthvh, 2025-07-16:
> I haven't been able to replicate this on v2.6.29 of the Android app, but can confirm that the bug is present on v2.6.10 of the iOS app. That node has an ID of `!mesh_tips`, and a node number of `4`.

Is this a Linux node?

jp-bennett, 2025-07-23:
Have a game plan to address the weirdness that results from changeable node IDs. Hopefully coming soon. (I have way too many ideas coming soon)

shalberd, 2025-10-05:
@thebentern @jp-bennett posting here since the issue seems similar to what I observe with another node:

I have an issue with a user node id shown in the conversation window of the main channel is different than the user id shown when clicking on node details from node list.
What is weird: I search for the id as shown in conversation window from the node list, then I get the node, with user long and short name, but then the user long and short name is displayed there, but not in main conversation window. It does not matter how many times I delete the node, the issue persists. What is causing this?

[image]

[image]

[image]

[image]

closed 2025-11-16 by DaneEvans