meshtastic/Meshtastic-Android #2411: [Feature Request]: Dark mode OSM map

shortwavesurfer2009, 2025-07-10:
### Tell us your idea.

It would be really useful to have dark mode for the open street map that would toggle on if the system dark mode was enabled and toggle off when it was disabled.

Currently the only way to get a dark map is to use the satellite view.

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

closed 2025-07-10 by shortwavesurfer2009

jamesarich, 2025-07-10:
> Oh, alright then. I refuse to use anything other than f-droid, so I'll just deal with it.

happy to accept PRs for osmdroid - just not going to plan to put effort towards it as a project priority since it's a dead dependency.

jamesarich, 2025-07-10:
https://github.com/maplibre/maplibre-native this is a potential alternative i found today, a privacy centric fork of mapbox (before they started collecting user data) - but haven't evaluated it against fdroid policies