meshtastic/Meshtastic-Android #2469: [Bug]: Unable to edit Public Key

bquinlan, 2025-07-20:
### Contact Details

rdq@aslanrefuge.org

### What happened?

I am now longer able to edit or clear the Public Key field.  The Private Key field still works correctly, but the Public Key field does not appear to be editable.

### App Version

2.6.30

### Phone

Samsung Galaxy S23 Ultra running Android 15.

### Device

Heltec T114 with GPS

### Firmware

2.7.2

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

jamesarich, 2025-07-20:
public key is derived from private - and is not editable by intention

closed 2025-07-20 by jamesarich

b8b8, 2025-07-20:
when you reboot the node after copying back in the private key, the public key will automatically regenerate.

bquinlan, 2025-07-21:
Thank you!  I looked, but couldn't find anything a out it.