meshtastic/Meshtastic-Android #2523: [Bug]: 2.6.31 - Environment metrics showing Soil entry

mjvolante, 2025-07-25:
### What happened?

2.6.31 - Environment Metrics Log - App showing entries for Soil Moisture and Soil Temperature. Device Seeed Solar Node with Grove SHT31 temp & humidity sensor.

[image]

### App Version

2.6.31

### Phone

Pixel 7, Android 15

### Device

Seeed P1 Solar Node

### Firmware

2.6.11

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

glith, 2025-07-26:
Same issue here.

RCGV1, 2025-07-27:
In the iOS app you can modify the graph, adding this here as this may be a good recourse on how to implement a feature to not present values a user doesn't need.[image][image]

jamesarich, 2025-07-27:
the android charts aren't all in one @RCGV1 - that doesn't make sense.

PR #2556 (2025-07-30): Fix/2523 redundant soil metrics
Needs to be looked at carefully. 
A lot more changes than I'd have liked, 

Changes
- remove soil metrics when no data
- And humidity 
- make 0.0 show up when it's real data 
- make graph and legend reflect only present data 
- Pulled env metrics chart into own file to make detekt happy 

closes #2523 

needs #2571
closed 2025-07-30 by jamesarich