meshtastic/Meshtastic-Android #2528: [Bug]: it doesn't show the network coverage map

jekeam, 2025-07-25:
### Contact Details

@jekeam

### What happened?

it doesn't show the network coverage map, I tried the pre-release, and it's the same. 

[image]

Why can't I see the nodes and their coverage?

### App Version

2.6.30 (2.6.31 - prerelease)

### Phone

Oxygenos, android 13

### Device

OnePlus Nord 2

### Firmware

None

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

b8b8, 2025-07-25:
The screenshot shows you aren't connected to a node.

jekeam, 2025-07-27:
@DaneEvans Yes, I changed the scale and different cities, changed the types of Internet (home and 3g/4g), connected and disconnected VPN, also installed preelease, in my city it works for others. But I have not added any equipment yet, it seems that the card starts working only if equipment is added? How can I make logs?

jamesarich, 2025-07-27:
This app is intended to be used in conjunction with a device, you will not be able to participate in the mesh without one. Thankfully there are a lot of inexpensive options: https://meshtastic.org/docs/hardware/devices/

There's plenty of other documentation available (and our discord is full of helpful users) for any further questions you might have - check it out.

closed 2025-07-27 by jamesarich

jekeam, 2025-07-27:
I understand that I can't participate in the network without a device, but I would like to see which nodes are currently operating in my city in order to plan and order the equipment I need. However, I can't find an adequate map of the current node coverage anywhere.

jamesarich, 2025-07-27:
there may not _be_ one. internet facing maps are backed by mqtt and are all optional to participate in.

jamesarich, 2025-07-27:
the app won't show you that either way, it's not for that use case at all.
the app shows nodes on the map that the current connected device has seen.