meshtastic/Meshtastic-Android #2530: [Bug]: Set to Fahrenheit but still shows in Celsius scale but with a F°

versteckt89, 2025-07-26:
### What happened?

Set to Fahrenheit but still shows in Celsius scale but with a F°

[image]

### App Version

2.6.31

### Phone

Pixel 9 PRO XL, Android 16 Beta Q3

### Device

RAK19007, RAK4631

### Firmware

2.6.11

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

DaneEvans, 2025-07-27:
Which part are we looking at here? The graph or the text? What do you think it should read? 

Is it only soil temperature?

versteckt89, 2025-07-27:
The number representing the temperature. 

I have two screenshots here, one with the setting "Environmental Metrics use Fahrenheit" enabled and another with that setting disabled. 

Enabled
[image]

Disabled.
[image]

DaneEvans, 2025-07-28:
Seems to be working fine on main from a Simulator >  Heltek @2.7.2 > T1000E @2.7.2

@versteckt89 - Can you confirm that the issue persists through reboots? 
I'll try rolling back to the .31 release and see if I can replicate there. 

Other than that it looks like we're into weird edgecases. 
- Are you connected directly, what interface? 
- Do you have any other devices that replicate / or don't? 
- Are there any others on the network that you can look at?

PR #2546 (2025-07-29): Fix/2530 temp scaling
Closes #2530 

- Fahrenheit conversion was being done, but the soil temp copy overwrote the regular temp one. 

- Added testing
closed 2025-07-29 by jamesarich